flathub/org.freecadweb.FreeCAD #1: IfcOpenShell support

mdhill, 2018-11-09:
If I add 'date_time' to the boost libraries, IfcOpenShell will build from git source as the last module before freecad. FreeCAD doesn't need the whole installation, just...
- the contents of the src/ifcopenshell-python folder, byte compiled
- ifcopenshell_wrapper.py and _ifcopenshell_wrapper.so

...installed into site-packages. It's not clear to me where the *wrapper* files come from, because they're not generated by the C++ build, and byte compiling still needs to be done explicitly.

The python module is packaged by architecture, and as an alternative to building from source, it be installed like this (not valid because it isn't portable?):

  - name: ifcopenshell
    buildsystem: simple
    build-commands:
      - python2 -m compileall .
      - mkdir -p /app/lib/python2.7/site-packages/ifcopenshell
      - cp -R * /app/lib/python2.7/site-packages/ifcopenshell
    sources:
      - type: archive
        url: https://github.com/IfcOpenShell/IfcOpenShell/releases/download/v0.5.0-preview2/ifcopenshell-python27-master-9ad68db-linux64.zip
        sha256: 523ce6dc357eb6fd38274fbb97b6af88fe05828842dea9fff0ac3892e796b3af

First problem: before this is committed to the cache, flatpak-builder tells me "Removing stale python bytecode file files/lib/python2.7/site-packages/ifcopenshell/geom/occ_utils.pyc" for each one I just generated. I checked... they're gone.

On the bright side: the resulting FreeCAD recognizes that IfcOpenShell is installed... no more "IfcOpenShell not found" crash when trying to open or import an IFC file.

Second problem: when I open or import an IFC file, nothing comes into the tree or the 3D view, and immediately the console says "Finished importing." On a working installation it generally takes a while, and stuff gets imported.

mdhill, 2018-11-09:
Wiki page:

[https://www.freecadweb.org/wiki/Import/Export_IFC_-_compiling_IfcOpenShell/en](url)

eszlari, 2018-11-09:
fixed: https://github.com/flathub/org.freecadweb.FreeCAD/commit/d55e932cd8c9faf079894ed2b3c02d2315b92349

build: https://flathub.org/builds/#/builders/2/builds/6003

mdhill, 2018-11-09:
@eszlari, thanks. It works here.

closed 2018-11-10 by eszlari