flathub/org.freecadweb.FreeCAD #7: Can't boot using desktop file

peteruithoven, 2019-04-17:
I noticed that I can't boot FreeCAD successfully from my application menu. This while I can successfully boot it by running:
```
flatpak run org.freecadweb.FreeCAD
```

The Exec from the .desktop file:
```
/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=freecad --file-forwarding org.freecadweb.FreeCAD @@ %F @@
```
When I run this in the Terminal I get:
```
bwrap: execvp freecad: No such file or directory
```

I noticed that when I remove the `--command=freecad` it does boot successfully. 
--command=freecad

eszlari, 2019-04-17:
Here it's:
```
/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=FreeCAD --file-forwarding org.freecadweb.FreeCAD @@ %F @@
```

Could you run
```
flatpak update
```
It's seems you're using a outdated version.

peteruithoven, 2019-04-17:
Looks like the issue was that I had in some way created an overriding .desktop file when I renamed it to `FreeCAD (flatpak)`. 
When I removed that .desktop file it started using another one which does boot. (It contains `--command=FreeCAD`).

(There was no update and re-installing the FreeCAD flatpak didn't help)

Thanks for the quick reply.

closed 2019-04-17 by peteruithoven

eszlari, 2019-04-17:
You must have created that overriding .desktop file with an older version, because the executable was indeed named `freecad` until this commit: https://github.com/flathub/org.freecadweb.FreeCAD/commit/53417a5cee829ef165b8f6b02db0e14d014afcbc