flathub/org.freecadweb.FreeCAD #9: Program received signal SIGSEGV, Segmentation fault.

ouening, 2019-06-11:
OS: Ubuntu18.04

After I installed FreeCAD using flatpak and started the application, I got errors below:
`Gtk-Message: 11:27:24.604: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:27:24.604: Failed to load module "canberra-gtk-module"
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x385e0) [0x7fc4740035e0]
#1  /usr/lib/x86_64-linux-gnu/GL/nvidia-418-56/lib/libnvidia-glcore.so.418.56(+0xdcd0ba) [0x7fc466a000ba]
#2  /usr/lib/x86_64-linux-gnu/GL/nvidia-418-56/lib/libnvidia-glcore.so.418.56(+0xdcec74) [0x7fc466a01c74]
#3  /usr/lib/x86_64-linux-gnu/GL/nvidia-418-56/lib/libGLX_nvidia.so.0(+0x52ad1) [0x7fc467af5ad1]
`
Can someone fix it?

eszlari, 2019-06-11:
Please provide the output of:

```
$ flatpak --version
$ flatpak --gl-drivers
$ flatpak list -a
```

Do other apps work (e.g. Krita, Blender, Kdenlive) ?

ouening, 2019-06-13:
Hi, the output of above commands lists below:

```
flatpak --version                                                           ⏎ [12d23h28m] ✹
Flatpak 1.0.8

flatpak --gl-drivers                                                        ⏎ [12d23h29m] ✹
nvidia-418-56
default
host

flatpak list -a                                                               [12d23h29m] ✹
Ref                                                      Options                                                                         
com.github.eudaldgr.elements/x86_64/stable               system,current,eol=Elements is no longer available as a Flatpak package.        
com.github.hopsan.Hopsan/x86_64/stable                   system,current                                                                  
com.gitlab.bitseater.meteo/x86_64/stable                 system,current                                                                  
com.leinardi.gwe/x86_64/stable                           system,current                                                                  
io.github.markummitchell.Engauge_Digitizer/x86_64/stable system,current                                                                  
net.oz9aec.Gpredict/x86_64/stable                        system,current                                                                  
org.freecadweb.FreeCAD/x86_64/stable                     system,current                                                                  
org.inkscape.Inkscape/x86_64/stable                      system,current                                                                  
org.kde.kdenlive/x86_64/stable                           system,current                                                                  
org.octave.Octave/x86_64/stable                          system,current                                                                  
org.remmina.Remmina/x86_64/stable                        system,current                                                                  
com.github.eudaldgr.elements.Locale/x86_64/stable        system,runtime,partial,eol=Elements is no longer available as a Flatpak package.
com.gitlab.bitseater.meteo.Locale/x86_64/stable          system,runtime,partial                                                          
net.oz9aec.Gpredict.Locale/x86_64/stable                 system,runtime,partial                                                          
org.freedesktop.Platform.GL.nvidia-410-78/x86_64/1.4     system,runtime                                                                  
org.freedesktop.Platform.GL.nvidia-418-56/x86_64/1.4     system,runtime                                                                  
org.freedesktop.Platform.Locale/x86_64/18.08             system,runtime,partial                                                          
org.freedesktop.Platform.ffmpeg/x86_64/1.6               system,runtime                                                                  
org.freedesktop.Platform.html5-codecs/x86_64/18.08       system,runtime                                                                  
org.freedesktop.Platform/x86_64/18.08                    system,runtime                                                                  
org.gnome.Platform.Locale/x86_64/3.28                    system,runtime,partial                                                          
org.gnome.Platform.Locale/x86_64/3.30                    system,runtime,partial                                                          
org.gnome.Platform.Locale/x86_64/3.32                    system,runtime,partial                                                          
org.gnome.Platform/x86_64/3.28                           system,runtime                                                                  
org.gnome.Platform/x86_64/3.30                           system,runtime                                                                  
org.gnome.Platform/x86_64/3.32                           system,runtime                                                                  
org.gtk.Gtk3theme.Ambiance/x86_64/3.22                   system,runtime                                                                  
org.inkscape.Inkscape.Locale/x86_64/stable               system,runtime                                                                  
org.kde.Platform.Locale/x86_64/5.11                      system,runtime,partial                                                          
org.kde.Platform.Locale/x86_64/5.12                      system,runtime,partial                                                          
org.kde.Platform/x86_64/5.11                             system,runtime                                                                  
org.kde.Platform/x86_64/5.12                             system,runtime                                                                  
org.kde.Sdk.Locale/x86_64/5.12                           system,runtime,partial                                                          
org.kde.Sdk/x86_64/5.12                                  system,runtime                                                                  
org.kde.kdenlive.Locale/x86_64/stable                    system,runtime,partial                                                          
org.remmina.Remmina.Locale/x86_64/stable                 system,runtime,partial                                                          

```
Kdenlive installed from flatpak has no problem.

eszlari, 2019-07-17:
Seems to be a bug in the Nvidia driver. Could you try a newer version?

Newer Nvidia drivers are now provided for LTS releases:
https://www.omgubuntu.co.uk/2019/07/install-nvidia-driver-update-ubuntu-its

eszlari, 2020-05-03:
Closing this, as it seems to be a driver bug.

closed 2020-05-03 by eszlari