flathub/org.freecadweb.FreeCAD #11: Crash opening an stl file

yaneti, 2019-08-17:
Hi,  fedora rawhide user here.
```
$ flatpak info org.freecadweb.FreeCAD  

FreeCAD - An open source parametric 3D CAD modeler

          ID: org.freecadweb.FreeCAD
         Ref: app/org.freecadweb.FreeCAD/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 0.18.3
     License: LGPL-2.1
      Origin: flathub
  Collection: org.flathub.Stable
Installation: user
   Installed: 744.4 MB
     Runtime: org.kde.Platform/x86_64/5.12
         Sdk: org.kde.Sdk/x86_64/5.12

      Commit: 2b74926f734bf59841fee5cfc3eb888fc0432babda7911937ce9494d1b65fe71
      Parent: e12b47691f78b65d0fc5b33bba0f95b0d2420e473da99b2fd0874687e0955838
     Subject: v0.18.3 (#10) (58c1e9fe)
        Date: 2019-07-18 09:14:56 +0000
```
I get an immediate crash trying to open the the attached stl file 
[prt0001_49.stl.gz](https://github.com/flathub/org.freecadweb.FreeCAD/files/3511930/prt0001_49.stl.gz)

```
$ flatpak run org.freecadweb.FreeCAD  ~/temp/prt0001_49.stl 
FreeCAD 0.18, Libs: 0.18R14555 (Git shallow)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x385e0) [0x7fa15c6485e0]
#1  /usr/lib/x86_64-linux-gnu/libX11.so.6(XDefaultScreenOfDisplay+0x4) [0x7fa158b4a034]
#2  /app/lib/libCoin.so.4.0.0(+0x4ac84d) [0x7fa15c19d84d]
#3  /app/lib/libCoin.so.4.0.0(glxglue_init+0x3f) [0x7fa15c19e15f]
#4  /app/lib/libCoin.so.4.0.0(cc_glglue_instance+0x233) [0x7fa15c196ce3]
#5  0x7fa15c0236b8 in SoGLRenderActionP::isDirectRendering(SoState const*) const from /app/lib/libCoin.so.4.0.0+0x48
#6  0x7fa15c025a80 in SoGLRenderActionP::render(SoNode*) from /app/lib/libCoin.so.4.0.0+0xc0
#7  0x7fa15c025cc3 in SoGLRenderAction::beginTraversal(SoNode*) from /app/lib/libCoin.so.4.0.0+0xa3
#8  0x7fa15c01ed12 in SoAction::apply(SoNode*) from /app/lib/libCoin.so.4.0.0+0x372
#9  0x7fa15e36c88d in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /app/freecad/lib/libFreeCADGui.so+0x2d
#10  0x7fa15e407435 in Gui::View3DInventorViewer::renderScene() from /app/freecad/lib/libFreeCADGui.so+0x145
#11  0x7fa15e3d62e9 in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /app/freecad/lib/libFreeCADGui.so+0x89
#12  0x7fa15e3ddf81 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /app/freecad/lib/libFreeCADGui.so+0x21
#13  0x7fa15d101db8 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d8
#14  0x7fa15d1a6a5e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1e
#15  0x7fa15d3f2cf3 in QGraphicsView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x163
#16  0x7fa15e3d64ab in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x9b
#17  0x7fa15cc1f23b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#18  0x7fa15d0c25c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x71
#19  0x7fa15d0c9980 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#20  0x7fa15e1c3110 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#21  0x7fa15cc1f529 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#22  0x7fa15d0faa45 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x35
#23  0x7fa15d0fb307 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x867
#24  0x7fa15d0fbf01 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4d1
#25  0x7fa15d0faf8c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4ec
#26  0x7fa15d0fbf01 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4d1
#27  0x7fa15d0faf8c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4ec
#28  0x7fa15d0fbf01 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4d1
#29  0x7fa15d0faf8c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4ec
[... 18 lines cut ...]
#48  0x7fa15d0faf8c in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x4ec
#49  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x18937c) [0x7fa15d0d237c]
#50  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1895b4) [0x7fa15d0d25b4]
#51  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d5ae4) [0x7fa15d11eae4]
#52  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d659e) [0x7fa15d11f59e]
#53  0x7fa15d0c25d1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x81
#54  0x7fa15d0c9980 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x210
#55  0x7fa15e1c3110 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#56  0x7fa15cc1f529 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x179
#57  0x7fa15af88ed6 in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x96
#58  0x7fa15af8915d in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x1bd
#59  0x7fa15af64b8b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#60  /usr/lib/x86_64-linux-gnu/libQt5WaylandClient.so.5(+0x93fc0) [0x7fa156f12fc0]
#61  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2ae) [0x7fa158cec4ae]
#62  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52748) [0x7fa158cec748]
#63  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7fa158cec7e0]
#64  0x7fa15cc731f3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x63
#65  0x7fa15cc1e12b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#66  0x7fa15cc26112 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#67  0x7fa15e15c630 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x14f0
#68  /app/freecad/bin/FreeCAD(main+0x6b7) [0x55f5a1cbbc27]
#69  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fa15c6343e3]
#70  /app/freecad/bin/FreeCAD(_start+0x2e) [0x55f5a1cbbf9e]

```

eszlari, 2019-08-17:
Doesn't crash here (Kubuntu 19.04), but the 3D model view is all black.

jacknlliu, 2019-08-21:
Also crashed with Fedora 30.

eszlari, 2019-08-22:
Are you running a Wayland or X11 session?

Does it crash when you start the flatpak with:
```
$ flatpak run --env=QT_QPA_PLATFORM=xcb org.freecadweb.FreeCAD
```

jacknlliu, 2019-08-23:
@eszlari it works! Should we always use this option to run it?

eszlari, 2019-08-23:
@jacknlliu I made it the default in commit https://github.com/flathub/org.freecadweb.FreeCAD/commit/82adfb6f48ca590834447971896b49b93755d6bc, should work out of the box, when the build is done.

yaneti, 2019-08-23:
Also under wayland, and the xcb trick fixes it here too.

eszlari, 2019-08-23:
Ok, then I'm closing this.

When the build is finished, it will be fixed for everybody:
https://flathub.org/builds/#/builders/32/builds/6930

closed 2019-08-23 by eszlari