flathub/org.freecadweb.FreeCAD #16: Is there a badge that i can use to track the flatpak version?

luzpaz, 2019-10-11:
Tracking FreeCAD packages across different distros over at this [thread](https://forum.freecadweb.org/viewtopic.php?f=42&t=36818) in the FreeCAD forum. Is there a badge that can track this package ?

TingPing, 2019-10-11:
Not currently.

luzpaz, 2019-10-11:
what about a script that can tell me what the latest revision or version this flatpak is ?

TingPing, 2019-10-11:
`>flatpak remote-info flathub org.freecadweb.FreeCAD | grep Version`

I believe the flathub.org website does have a JSON API but I'm not familiar with it, I'll look into it.

luzpaz, 2019-10-11:
No version number at all

```
        ID: org.freecadweb.FreeCAD
       Ref: app/org.freecadweb.FreeCAD/x86_64/stable
      Arch: x86_64
    Branch: stable
Collection: org.flathub.Stable
  Download: 248.4?MB
 Installed: 746.2?MB
   Runtime: org.kde.Platform/x86_64/5.12

       Sdk: org.kde.Sdk/x86_64/5.12
    Commit: b0c6933e8e98ffd6f79e8495e83dc34de2c62cd43ea11b1b175bfca74d06203e
    Parent: 9d68f0e5192e227d6492c2ec952575e96be49a453b7f6f08561069ed5c6ea9e3
   Subject: add pycollada to enable collada export (#13) (2829179e)
      Date: 2019-09-18 02:25:09 +0000
```

TingPing, 2019-10-11:
You have an old version of `flatpak`.

luzpaz, 2019-10-11:
```bash
$ flatpak --version
Flatpak 1.5.0
```

TingPing, 2019-10-12:
Weird...

```
>flatpak remote-info flathub org.freecadweb.FreeCAD                                                                            (wip/tingping/localhost-is-local|…35)

FreeCAD - An open source parametric 3D CAD modeler

        ID: org.freecadweb.FreeCAD
       Ref: app/org.freecadweb.FreeCAD/x86_64/stable
      Arch: x86_64
    Branch: stable
   Version: 0.18.3
   License: LGPL-2.1
Collection: org.flathub.Stable
  Download: 248.4 MB
 Installed: 746.2 MB
   Runtime: org.kde.Platform/x86_64/5.12

       Sdk: org.kde.Sdk/x86_64/5.12
    Commit: b0c6933e8e98ffd6f79e8495e83dc34de2c62cd43ea11b1b175bfca74d06203e
    Parent: 9d68f0e5192e227d6492c2ec952575e96be49a453b7f6f08561069ed5c6ea9e3
   Subject: add pycollada to enable collada export (#13) (2829179e)
      Date: 2019-09-18 02:25:09 +0000
```

on 1.4.3

TingPing, 2019-10-12:
Maybe `flatpak update --appstream`

luzpaz, 2019-10-12:
Yes,  that worked. I ran that and then re-ran the original command and I see a version now. Thanks!

closed 2019-10-12 by luzpaz