flathub/org.freecadweb.FreeCAD #18: No support for space navigator input device

dbrgn, 2019-12-13:
I have a SpaceNavigator 3D mouse. In Blender it works just fine, but in FreeCAD from FlatHub it doesn't seem to work. Additionally, I don't see the "Spaceball" tabs in the settings that should be there according to https://www.freecadweb.org/wiki/3Dconnexion_input_devices#Setting_up_FreeCAD.

Is this a Flatpak permission issue? It seems that libspnav is being built by the package...

dbrgn, 2019-12-13:
Starting the program using `--device=all` doesn't seem to help either:

    flatpak run --branch=stable --arch=x86_64 --command=FreeCAD --file-forwarding --device=all org.freecadweb.FreeCAD

dbrgn, 2019-12-13:
Probably related: https://github.com/flathub/org.blender.Blender/issues/18

eszlari, 2020-04-20:
https://github.com/flatpak/flatpak/issues/3541

adrianinsaval, 2022-02-06:
posting in case it's relevant, socket path seems to be configurable now: https://github.com/FreeSpacenav/libspnav/releases/tag/v0.3
is it possible to use this to make it work in the flatpak?

PR #55 (2022-02-11): Update libspnav and pass along its socket
This will allow the use of space navigator input devices in FreeCAD,
verified with my Magellan Plus.

As /var/run is a symlink to /run on modern systems, this will not
require any change in spacenavd configuration either, and it won't have
any difference to users without such an input device - as it'll simply
ignore the passthrough if the socket does not exist when starting the
application.

Fixes #18
commit be284e63 (2022-02-11): Update libspnav and pass along its socket

This will allow the use of space navigator input devices in FreeCAD,
verified with my Magellan Plus.

As /var/run is a symlink to /run on modern systems, this will not
require any change in spacenavd configuration either, and it won't have
any difference to users without such an input device - as it'll simply
ignore the passthrough if the socket does not exist when starting the
application.

Fixes #18
closed 2022-02-11 by hfiguiere