flathub/org.freecadweb.FreeCAD #19: Strange color in left menu

Inglebard, 2020-01-16:
Hi,

I notice some strange background colors. It appears on some gradients too.
I notice on native build and appimage don't have this issue. Instead of be black, it's gray.

[image]

eszlari, 2020-04-20:
I cannot reproduce this. Do you still see it with the runtime update to 5.14 (https://github.com/flathub/org.freecadweb.FreeCAD/commit/d4130167c1a9b8439d5e3720e3b53889a1b857d2)?

Inglebard, 2020-04-20:
Hi,

Yes, the issue is still present.
For more information, I am on Archlinux and use nvidia driver.
Flatpak gl drivers : nvidia-440-82.

Inglebard, 2020-05-13:
Hi,

For some reason, the problem is no more present.

eszlari, 2020-05-13:
Ok.

closed 2020-05-13 by eszlari