flathub/org.freecadweb.FreeCAD #22: Enable access to Gnome VFS mounts

haasalex, 2020-05-01:
When I run FreeCAD using Flatpak, I can not access any of the network shares mounted using Nautilus. There is no access to the hosts `/run/user/$UID/gvfs/`directory.

If I understand https://github.com/flatpak/flatpak/issues/2900#issuecomment-492493307 correctly, adding this line
```
  - --filesystem=xdg-run/gvfs # needed to make GnomeVFS accessible
```
would fix the issue.

eszlari, 2020-05-01:
You can test this yourself easily! Just start FreeCAD with:

```
$ flatpak run --filesystem=xdg-run/gvfs org.freecadweb.FreeCAD
```

haasalex, 2020-05-02:
I did not realize that this is a plain runtime parameter. I might have guessed it if the parameters for `command` would be called `parameters`, not `finish-args`...

I did try it and can confirm: It does work as expected. I can use GVFS mounts with FreeCAD Flatpak this way.

I #23

closed 2020-05-02 by haasalex

eszlari, 2020-05-02:
Using search shows, that other apps also set two dbus permissions additionally:
```
--filesystem=xdg-run/gvfs
--talk-name=org.gtk.vfs
--talk-name=org.gtk.vfs.*
```
Do they make a difference on your system?

haasalex, 2020-05-02:
I might get it wrong, but to me it seems as if the DBUS org.gtk.vfs stuff is only useful if your application is using GIO (and GLib for that matter) like GTK does it.

FreeCAD is using qt all the way, so there is no need to have it.

Without this settings, on my machine, FreeCAD can read and write files. When I open a FCStd file in Nautilus, FreeCAD does open it. So I am glad with just the filesystem setting.

eszlari, 2020-05-14:
@haasalex What flatpak version did you experience this with?

haasalex, 2020-05-19:
@eszlari It was flatpak 1.7.2. Now I am curious why it would matter?

eszlari, 2020-05-19:
https://github.com/flathub/org.kde.krita/issues/21#issuecomment-628564334