flathub/org.freecadweb.FreeCAD #24: Add openSCAD

Emreu, 2020-08-18:
Please add openSCAD binary and libs with dependencies to bundle. Right now it seems impossible to import .scad files since FreeCAD require openscad binary which is using other libraries. I'm not proficient with flatpak and wasn't able to find any solution with standard tools and options.
May be this app repo might be of help: https://github.com/flathub/org.openscad.OpenSCAD

hfiguiere, 2020-10-13:
Actually #30 is needed for this to work.

hfiguiere, 2020-10-14:
Working on something using the openSCAD flatpak.

closed 2020-10-18 by eszlari

akhilman, 2024-02-27:
There is work to provide a better way to run application from another flatpak app.
https://github.com/flatpak/xdg-desktop-portal/issues/283

luzpaz, 2024-08-30:
Note: https://github.com/FreeCAD/FreeCAD/issues/15926#issuecomment-2295373765

PR #32 (2020-10-18): Support openscad import properly
Closes #30 , Closes #24

You need the openscad flatpak.
It goes .scad -> .csg -> FreeCAD.
So if importing a .scad works, importing a .csg works.