flathub/org.freecadweb.FreeCAD #25: Missing FreeCAD help files

exponentactivity, 2020-09-02:
When i use the "What Is This?" feature or open Help i get a dialog saying `FreeCAD help files not found (/app/freecad/doc/freecad.qhc). You might need to install the FreeCAD documentation package.`

skewty, 2021-05-23:
Quickly coming up a year and this is still an issue.  Is help required to get this working?

hfiguiere, 2021-05-23:
are you offering?

skewty, 2021-05-23:
I am spread far too thin at this time to take on the learning curve required to be less than an annoying with regards to Flatpak bundling at this time.  I am a seasoned python dev though so I may be useful in getting issue forty-one (don't want to link issues) closed though.

Additional but relevant context:
FreeCAD in linux + wayland is not really an option for me at this time.   
See this issue: https://tracker.freecadweb.org/view.php?id=4594
I am not going to hold my breath as the Wayland market share is very small and rife with venom and spite. :)

PR #43 (2021-05-29): Create manual extension (optional install)
- Closes #25

After this the manual can be installed as a separate package. It's 200+MB.
commit 3c43c765 (2021-05-29): Create manual extension (optional install)

- Closes #25
closed 2021-05-29 by hfiguiere