flathub/org.freecadweb.FreeCAD #46: aarch64 support

hfiguiere, 2021-12-04:
This is the issue for aarch64 support.

tl;dr it doesn't work.

It builds, thanks to some of the changes in #35.

The KDE sdk issue https://invent.kde.org/packaging/flatpak-kde-runtime/-/issues/19

luzpaz, 2021-12-04:
Release notes have been updated to link to #46 (this issue)   
Thanks @hfiguiere!

hfiguiere, 2021-12-09:
Related:

https://github.com/coin3d/coin/issues/449

luzpaz, 2022-01-07:
Upstream Coin3d just provided a fix see https://github.com/coin3d/coin/pull/470

hfiguiere, 2022-01-10:
not it didn't. EGL is not GLES. EGL is akin to GLX, but for Wayland. 

What this would allow is FreeCAD working on Wayland.

luzpaz, 2022-01-10:
:disappointed:

hfiguiere, 2022-04-11:
I a new plot twist, the current 0.20 fails to build now without Qt5 GLES

ok so now 0.20 doesn't build with GLES anymore:
```
[3274/4252] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o
FAILED: src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o 
/usr/bin/ccache /usr/bin/c++ -DBUILD_ADDONMGR -DCMAKE_BUILD_TYPE=\"RelWithDebInfo\" -DFreeCADGui_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_GLES2 -DHAVE_PYSIDE2 -DHAVE_SHIBOKEN2 -DHAVE_TBB -DHAVE_VTK -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DSPNAV_FOUND -DVTK_OPENGL2_BACKEND -D_OCC64 -DvtkDomainsChemistry_AUTOINIT="1(vtkDomainsChemistryOpenGL2)" -DvtkIOExport_AUTOINIT="2(vtkIOExportOpenGL2,vtkIOExportPDF)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL2)" -DvtkRenderingCore_AUTOINIT="3(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingOpenGL2)" -DvtkRenderingOpenGL2_AUTOINIT="1(vtkRenderingGL2PSOpenGL2)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL2)" -I/run/build/FreeCAD/_flatpak_build/src/Gui/FreeCADGui_autogen/include -I/run/build/FreeCAD/_flatpak_build -I/run/build/FreeCAD/_flatpak_build/src -I/run/build/FreeCAD/src -I/run/build/FreeCAD/src/Gui -I/run/build/FreeCAD/src/Gui/Quarter -I/run/build/FreeCAD/_flatpak_build/src/Gui -I/run/build/FreeCAD/src/Gui/.. -I/run/build/FreeCAD/_flatpak_build/src/Gui/.. -I/run/build/FreeCAD/_flatpak_build/src/Gui/Language -I/run/build/FreeCAD/_flatpak_build/src/Gui/propertyeditor -I/run/build/FreeCAD/_flatpak_build/src/Gui/TaskView -I/run/build/FreeCAD/_flatpak_build/src/Gui/Quarter -I/run/build/FreeCAD/_flatpak_build/src/Gui/DAGView -I/app/include/eigen3 -I/usr/include/python3.9 -I/app/include/shiboken2 -I/app/include/PySide2 -I/app/include/PySide2/QtCore -I/app/include/PySide2/QtGui -I/app/include/PySide2/QtWidgets -isystem /usr/include/QtCore -isystem /usr/lib/mkspecs/linux-g++ -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtOpenGL -isystem /usr/include/QtPrintSupport -isystem /usr/include/QtSvg -isystem /usr/include/QtNetwork -isystem /usr/include/QtUiTools -isystem /usr/include/QtXml -Wall -Wextra -Wno-write-strings -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -O2 -g -DNDEBUG -fPIC -pthread -fPIC -std=gnu++17 -Wno-deprecated-copy -MD -MT src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o -MF src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o.d -o src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o -c /run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp: In member function ‘virtual void SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*)’:
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:869:18: error: ‘GL_PROJECTION’ was not declared in this scope; did you mean ‘GL_LOCATION’?
  869 |     glMatrixMode(GL_PROJECTION);
      |                  ^~~~~~~~~~~~~
      |                  GL_LOCATION
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:869:5: error: ‘glMatrixMode’ was not declared in this scope
  869 |     glMatrixMode(GL_PROJECTION);
      |     ^~~~~~~~~~~~
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:910:5: error: ‘glPushAttrib’ was not declared in this scope
  910 |     glPushAttrib(GL_MULTISAMPLE_BIT_EXT);
      |     ^~~~~~~~~~~~
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:912:5: error: ‘glPopAttrib’ was not declared in this scope
  912 |     glPopAttrib();
      |     ^~~~~~~~~~~
[3275/4252] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/CallTips.cpp.o
[3276/4252] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidgetP.cpp.o
```

hfiguiere, 2022-04-11:
This is with commit d0485c6bef of FreeCAD

luzpaz, 2022-04-16:
CC @wmayer since https://github.com/FreeCAD/FreeCAD/commit/d0485c6bef78cf84065cc2a27e6e21fd4e6ab33d is your commit

luzpaz, 2022-04-18:
@donovaly just sending you a heads up about this issue. (For context, this repo builds the FreeCAD flatpack)

hfiguiere, 2022-06-16:
New issue: the latest version of pyside2 seems to fail to build on the aarch64 SDK (GLES2)

hfiguiere, 2022-06-16:
Downgrading pyside2 works for aarch64.

Now FreeCAD fails to build:
```
[3291/4266] Building CXX object src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o
FAILED: src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o 
/usr/bin/ccache /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILD_ADDONMGR -DCMAKE_BUILD_TYPE=\"RelWithDebInfo\" -DFreeCADGui_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_GLES2_EXT -DHAVE_OPENGL_EXT -DHAVE_PYSIDE2 -DHAVE_RAPIDJSON -DHAVE_SHIBOKEN2 -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DHAVE_XLIB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WIDGETS_LIB -DQT_XML_LIB -DSPNAV_FOUND -DVTK_OPENGL2_BACKEND -D_OCC64 -DvtkDomainsChemistry_AUTOINIT="1(vtkDomainsChemistryOpenGL2)" -DvtkIOExport_AUTOINIT="2(vtkIOExportOpenGL2,vtkIOExportPDF)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL2)" -DvtkRenderingCore_AUTOINIT="3(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingOpenGL2)" -DvtkRenderingOpenGL2_AUTOINIT="1(vtkRenderingGL2PSOpenGL2)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL2)" -I/run/build/FreeCAD/_flatpak_build/src/Gui/FreeCADGui_autogen/include -I/run/build/FreeCAD/_flatpak_build -I/run/build/FreeCAD/_flatpak_build/src -I/run/build/FreeCAD/src -I/run/build/FreeCAD/src/Gui -I/run/build/FreeCAD/src/Gui/Quarter -I/run/build/FreeCAD/_flatpak_build/src/Gui -I/run/build/FreeCAD/src/Gui/.. -I/run/build/FreeCAD/_flatpak_build/src/Gui/.. -I/run/build/FreeCAD/_flatpak_build/src/Gui/Language -I/run/build/FreeCAD/_flatpak_build/src/Gui/propertyeditor -I/run/build/FreeCAD/_flatpak_build/src/Gui/TaskView -I/run/build/FreeCAD/_flatpak_build/src/Gui/Quarter -I/run/build/FreeCAD/_flatpak_build/src/Gui/DAGView -I/app/include/eigen3 -I/usr/include/python3.9 -I/app/include/shiboken2 -I/app/include/PySide2 -I/app/include/PySide2/QtCore -I/app/include/PySide2/QtGui -I/app/include/PySide2/QtWidgets -isystem /usr/include/QtCore -isystem /usr/lib/mkspecs/linux-g++ -isystem /usr/include/QtWidgets -isystem /usr/include/QtGui -isystem /usr/include/QtOpenGL -isystem /usr/include/QtPrintSupport -isystem /usr/include/QtSvg -isystem /usr/include/QtNetwork -isystem /usr/include/QtUiTools -isystem /usr/include/QtXml -Wall -Wextra -Wno-write-strings -O2 -g -pipe -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -fasynchronous-unwind-tables -fstack-clash-protection -fno-omit-frame-pointer -O2 -g -DNDEBUG -fPIC -pthread -fPIC -std=gnu++17 -Wno-deprecated-copy -MD -MT src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o -MF src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o.d -o src/Gui/CMakeFiles/FreeCADGui.dir/Quarter/QuarterWidget.cpp.o -c /run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp: In member function ‘virtual void SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*)’:
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:869:18: error: ‘GL_PROJECTION’ was not declared in this scope; did you mean ‘GL_LOCATION’?
  869 |     glMatrixMode(GL_PROJECTION);
      |                  ^~~~~~~~~~~~~
      |                  GL_LOCATION
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:869:5: error: ‘glMatrixMode’ was not declared in this scope
  869 |     glMatrixMode(GL_PROJECTION);
      |     ^~~~~~~~~~~~
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:910:5: error: ‘glPushAttrib’ was not declared in this scope
  910 |     glPushAttrib(GL_MULTISAMPLE_BIT_EXT);
      |     ^~~~~~~~~~~~
/run/build/FreeCAD/src/Gui/Quarter/QuarterWidget.cpp:912:5: error: ‘glPopAttrib’ was not declared in this scope
  912 |     glPopAttrib();
      |     ^~~~~~~~~~~
```

THS-on, 2024-09-13:
@hfiguiere whats the current state of getting FreeCAD to run on ARM64?

We have interest in getting this working for the M1/M2 devices and I probably can spend some time on trying to get it to work.

hfiguiere, 2024-09-14:
pull requests are accepted.

Note that part of it is the runtime that doesnt support plain GL because Qt doesn't allow both compiled at the same time. (it works on RaspberryPi if I rebuild the SDK with "desktop" GL and that's 24 hours of build on RPi400)

Also I'm not sure how GL support is in freedesktop-sdk for that specific platform. Might just work.

adrianinsaval, 2024-09-16:
I'm working on making this use qt6, in theory that should enable it to work for both x86 and arm, with qt5 unfortunately it is not currently possible because kde's runtime prioritizes mobile devices so they only support OpenGL ES and not full OpenGL which is not enough for coin3d

THS-on, 2024-09-16:
The Asahi Linux folks have their own overlay for mesa, see: https://pagure.io/fedora-asahi/mesa-asahi-flatpak

@adrianinsaval good to know, I'll then wait for your changes before testing more in-depth.

hfiguiere, 2024-09-16:
Freedesktop-sdk include the asahi drivers on aarch64. So the problem remain Qt5.

hfiguiere, 2024-09-16:
they are "untested" according to the freedesktop-sdk maintainers. Probably because they don't have the hardware.

hfiguiere, 2024-09-26:
https://github.com/flathub/org.freecad.FreeCAD build it now.

closed 2024-09-26 by hfiguiere