flathub/org.freecadweb.FreeCAD #54: In preparation for the future: Qt6

luzpaz, 2022-01-27:
Even though v0.20 will still support min version of Qt 5.9 I just wanted to put on the radar that the team is discussing  support for [Qt6](https://github.com/FreeCAD/FreeCAD-Bundle/issues/90), FWIW

hfiguiere, 2022-01-27:
We'll keep an eye on it. Python 3.10 is coming in the August 2022 runtime update so we'll need to be cautious.

adrianinsaval, 2022-05-30:
main issue with python 3.10 that I can remember was pivy, version 0.6.7 is supposedly compatible with it so that's the first thing to update when 3.10 comes.

hfiguiere, 2022-06-08:
pivy 0.6.7 will be in the next build on beta.

hfiguiere, 2022-06-15:
we have to revert the pivy update: some tests fail.

adrianinsaval, 2022-06-22:
fix for pivy 0.6.7 error: https://github.com/FreeCAD/FreeCAD/commit/b85e9b1fe4 it will be in the 0.20.1 release (no date estimate for it) but feel free to use as a patch meanwhile.

adrianinsaval, 2022-08-27:
when updating to py3.10 pivy should be updated at least to the recently released 0.6.8: https://github.com/coin3d/pivy/releases/tag/0.6.8 it has an important fix for a crash: https://github.com/coin3d/pivy/issues/98

closed 2024-11-19 by adrianinsaval