flathub/org.freecadweb.FreeCAD #58: beta release in flathub

eoli3n, 2022-03-08:
could you please publish the 0.20 in beta branch ?

hfiguiere, 2022-03-09:
upstream doesn't do.

hfiguiere, 2022-03-17:
Closed with #60

closed 2022-03-17 by hfiguiere

eoli3n, 2022-03-17:
Great ! But how to install ?

eoli3n, 2022-03-17:
how is it possible, could you find the link you use for flathub-beta repository ?

kinghat, 2022-03-17:
> how is it possible, could you find the link you use for flathub-beta repository ?

i dunno because its whatever discover did behind the scenes, but the bot built it here: https://github.com/flathub/org.freecadweb.FreeCAD/pull/60#issuecomment-1070320424

> flatpak install --user https://dl.flathub.org/build-repo/79904/org.freecadweb.FreeCAD.flatpakref

hfiguiere, 2022-03-17:
but you will not get updates.

hfiguiere, 2022-03-17:
To use the beta repository:

https://discourse.flathub.org/t/how-to-use-flathub-beta/2111

PR #60 (2022-03-17): (beta) Weekly build 14 March
Closes #58 
Closes #52