flathub/org.freecadweb.FreeCAD #59: The Flatpak version of freecad doesn't allow you to open multiple freecad windows at the same time

K4LCIFER, 2022-03-14:
If I open a second Freecad window of the flatpak version, i just get a black window that doesn't respond; however, a regular freecad package will launch multiple windows just fine.

P.S. Please let me know if this sort of bug is better suited for an official Freecad bug repor rather than here, as I am unsure the proper place for this.

hfiguiere, 2022-03-17:
Not sure how you open a new window. I have multiple document tabs though, no problem.

K4LCIFER, 2022-03-17:
I just open a second instance of FreeCAD, nothing special. Either by clicking the desktop icon or by running it from the command line.

adrianinsaval, 2022-05-30:
I can run multiple instances of the flatpak without issues, this could be DE specific

hfiguiere, 2022-06-27:
I can reproduce this.

adrianinsaval, 2022-09-13:
I can now also reproduce this and I think I found the culprit, running from terminal I get this error:
```[2:31:0913/143925.943288:ERROR:address_tracker_linux.cc(214)] Could not bind NETLINK socket: Address already in use (98)```
and googling it I found this https://github.com/flathub/org.qutebrowser.qutebrowser/issues/19
seems to be a bug in chromium (which is used in qtwebengine which is used in startpage and addon manager)
A workaround for the time being is to run `flatpak run --command=sh org.freecadweb.FreeCAD` and then `FreeCAD`

can this be somehow patched in qtwebengine?

adrianinsaval, 2023-09-21:
I built the flatpak with the kde 6.5 runtime, the error I mentioned is no longer present, but additional instances of FreeCAD still freeze.

pathquester, 2024-02-25:
Same issue here, not able to open multiple instances concurrently.

adrianinsaval, 2024-10-30:
In the org.freecad.FreeCAD beta starting from command line and hitting ctrl+c at some yet unidentified part of the startup process allows running two instances at once, it sometimes work and sometimes the whole program is terminated presumably dependent on at what time you hit ctrl+c. This was the output in report view in one of my successful attempts:
```
12:20:16  Failed to create a file lock for the IPC: Interrupted system call
12:20:17  Unhandled unknown exception caught in when notifying observer.
12:20:17  pyException: Traceback (most recent call last):
  File "/app/freecad/Mod/Tux/NavigationIndicatorGui.py", line 63, in onChange
    setCurrent()
  File "/app/freecad/Mod/Tux/NavigationIndicatorGui.py", line 777, in setCurrent
    gStyle.blockSignals(True)
    ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'PySide6.QtGui.QActionGroup' object has no attribute 'blockSignals'
```