flathub/org.freecadweb.FreeCAD #64: Collada DAE export broken, rebuild with new pycollada package needed

cjue, 2022-03-30:
The latest freecad flatpak commit uses Python 3.9 and a broken version of the pycollada package, which results in this error during .dae export:

```
Traceback (most recent call last):
  File "/home/juelg/.FreeCAD/Mod/RobotExport/Export_cmd.py", line 57, in Activated
    self.exportGroup(doc.Export.Group, path + u"/base_link")
  File "/home/juelg/.FreeCAD/Mod/RobotExport/Export_cmd.py", line 28, in exportGroup
    importDAE.export(parts, basename + u".dae")
  File "/app/freecad/Mod/Arch/importDAE.py", line 327, in export
    colmesh.write(filename)
  File "/app/lib/python3.9/site-packages/collada/__init__.py", line 554, in write
    self.save()
  File "/app/lib/python3.9/site-packages/collada/__init__.py", line 524, in save
    o.save()
  File "/app/lib/python3.9/site-packages/collada/geometry.py", line 232, in save
    if src.xmlnode not in meshnode.getchildren():

'xml.etree.ElementTree.Element' object has no attribute 'getchildren'
```

The underlying problem has been fixed in pycollada 0.7.2:
https://github.com/pycollada/pycollada/commit/5f235ce8a14dd1c84163ea82dfe67e72dab829c1

To my understanding the pycollada version is not explicitly specified anywhere here or in the freecad repo, so a simple rebuild should fetch the corrected pycollada dependency and solve this issue.

hfiguiere, 2022-03-30:
yes the version is specified in the flatpak:
https://github.com/flathub/org.freecadweb.FreeCAD/blob/master/org.freecadweb.FreeCAD.yaml#L351

hfiguiere, 2022-03-30:
is this with 0.19 or with the 0.20-weekly that is on flathub-beta?

cjue, 2022-03-30:
This is with version 0.19.4 from org.flathub.Stable

cjue, 2022-03-30:
The older stable version 0.19.2 / commit 0fe57f718dd27cc4c83a5e844f5c2742021a6e3fdda28afff72548ea38e37bfd does not have this issue, from my guess because of a pre-3.9 python.

commit 1cbe78d4 (2022-03-31): Update pycollada to 0.7.2

Close #64
commit 9ac8f8f8 (2022-04-03): [beta] Update pycollada to 0.7.2

Close #64
closed 2022-03-31 by hfiguiere