flathub/org.freecadweb.FreeCAD #76: New stable version to be released soon, suggestions

adrianinsaval, 2022-05-27:
Hi, as you may know FreeCAD 0.20 will get released soon and I have some suggestion for the flatpak release

In the past we've had misunderstandings in the forum because the flatpak versions report their version number incorrectly due to the shallow cloning of the repo (#2), at least for the stable release this can be solved by using the source package instead of a shallow clone, this package is generated after preprocessing the version information so it will report it correctly. (see https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/freecad_source.tar.gz)

Also please see this (https://forum.freecadweb.org/viewtopic.php?f=10&t=66407&start=100#p597687)
> - In the past we suffered from some OCC crashes. thanks to @chennes some are now fixes via OCC 7.6.2. Therefore we should try to release FreeCAD 0.20 using OCC 7.6.2 if possible. If no, this won't be a showstopper.
> - our fellow contributor @tanderson69 (a.k.a. blobfish) fixed further OCC bugs. Therefore, if possible one should compile OCC using his OCC 7.6.2 release:
> https://gitlab.com/blobfish/occt/-/tree/V7_6_2_BF

could you please use the 7.6.2 occt from the linked repo instead of the 7.5 from debian? There are important fixes there

Also related to occt, could you compile with rapidjson support? This is necessary to enable gltf export see https://forum.freecadweb.org/viewtopic.php?f=10&t=56487&hilit=rapidjson&start=20 and https://dev.opencascade.org/content/building-linux-rapidjson

hfiguiere, 2022-05-27:
> In the past we've had misunderstandings in the forum because the flatpak versions report their version number incorrectly due to the shallow cloning of the repo (#2), at least for the stable release [...]

I don't see a way to use that tarball. I dosn't contain the source just some files. If only there was a tag on the main tree like in previous release cycles.

> Also please see this (https://forum.freecadweb.org/viewtopic.php?f=10&t=66407&start=100#p597687)

I'll see what I can do. Last update was a shit show because of TBB and wasted countless hours figuring it out. So I'm hesitant to change anything.

> could you please use the 7.6.2 occt from the linked repo instead of the 7.5 from debian? There are important fixes there
>
> Also related to occt, could you compile with rapidjson support? This is [...]

As for 0.20 it doesn't look in good shape PySide and the QtWebEngine get in the way: see issue #61 
It's already build on 'beta"

hfiguiere, 2022-05-27:
> I don't see a way to use that tarball. I dosn't contain the source just some files. If only there was a tag on the main tree like in previous release cycles.

I'll rephrase. 

You link to an unversioned tarball, I can NOT use that. Because using it mean the build will break when it changes  as an archive has an associated sha256.

As from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20beta1 there is no proper source tarball.

hfiguiere, 2022-05-27:
Also https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/freecad_version.txt that I could parse changes daily. So much for a weekly build.

adrianinsaval, 2022-05-27:
What do you mean just some files? It's a snapshot of the git repo, so it is the entire source code, note that it has to be the file freecad_source.tar.gz, the source.zip and source.tar.gz that GitHub autogenerates from de bundle page are the sources just for the bundle scripts. There is no tag for 0.20 because it's not released as stable yet but once it's out there will be tag and a file like that with the source code and the correct version info will be there.

I'll see if I can come up with a PR next week but I've never built a flatpak so if you can figure stuff out don't wait on me.

Thanks for all your work!

adrianinsaval, 2022-05-27:
Another option is to get the correct version is to apply a patch for src/Build/Version.h.cmake with the correct info and keep the current system of making a shallow clone of a specific commit

adrianinsaval, 2022-05-27:
Sorry I responded to your first comment and only now see the other two, I was not suggesting to use that specific tarball but the equivalent stable version of that once it's out, I guess the one available at https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.19.4 would have been a better example.
And you are right, unfortunately the "weekly" release is incorrectly tagged, the binaries get more frequent updates now for heavy testing before release and the source tarball is automatically updated every 4h all the time (since it doesn't require much compute time)

hfiguiere, 2022-05-27:
There was a "beta1" announced but nothing. So....

adrianinsaval, 2022-05-27:
That beta release is mostly for the Windows installer as that one is done all manually unlike the rest.
@looooo could you upload a static freecad_source.tar.gz to the beta1 release in the bundle repo to use it on the flatpak?

looooo, 2022-05-27:
@adrianinsaval currently I don't see a `0.20_beta` tag. I guess it's best to still use the master branch for testing. OT: Regarding occt7.6: The conda-bundles will still use occt7.5. Some libraries (mainly IfcOpenShell and pythonocccore) are not yet ported to occt7.6.

adrianinsaval, 2022-05-27:
I was talking about this release: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20beta1 whatever the current snapshot that is up now will do in order to test before the release.
Regarding occ 7.6 I guess that can wait but it seems weird that the Windows release uses it without issues

hfiguiere, 2022-05-27:
we don't and can't use conda. it's an *offline* build.

hfiguiere, 2022-05-27:
As predicted the build fail with the new opencascade.

looooo, 2022-05-27:
> I was talking about this release: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.20beta1 whatever the current snapshot that is up now will do in order to test before the release.
> Regarding occ 7.6 I guess that can wait but it seems weird that the Windows release uses it without issues

Anyway I think it would be best to create a 0.20_beta as soon as possible. Otherwise we add additional confusion. beta, alpha, release... Who can create a beta tag for the FreeCAD-repo? @yorikvanhavre @wwmayer @channes

adrianinsaval, 2022-05-27:
@ppd if i'm not mistaken the snap packages are using occ 7.6 for some time, did you have to do some workaround?

donovaly, 2022-05-27:
> Anyway I think it would be best to create a 0.20_beta as soon as possible.

Done: https://github.com/FreeCAD/FreeCAD/releases/tag/0.20beta1

ppd, 2022-05-27:
@adrianinsaval No, but I don't think we have ifcopenshell & PythonOCC in the snap (yet, at least).

hfiguiere, 2022-05-28:
I have #78 for the rapidjson dependency.

Same changes ported to #75 that will also have the beta1 tagged (for the beta branch)

Let's see how the builds go.

adrianinsaval, 2022-05-28:
great, so opencascade compile options is done, that leaves us

- [ ] opencascade 7.6.2 - needs updates on ifcopenshell
ifcopenshell 0.7 is supposedly compatbile with 7.6 but has some extra dependencies https://github.com/conda-forge/ifcopenshell-feedstock/pull/20#issuecomment-1087294688 for some reason the build for conda doesn't find gmp and fails, don't know if flatpak would have the same issue.
- [ ] correct versioning for stable release
source package with version info is now available for beta1 https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/0.20beta1/freecad_source.tar.gz if this is used instead of the shallow clone it will have correct version info, once stable is out there should be a similar source tarball.
- [ ] correct versioning for future beta versions
I'll work on a script to generate a patch for src/Build/Version.h.cmake with the info, this could then be automated to update the beta periodically or could be run manually to update it.

adrianinsaval, 2022-05-28:
if #80 goes well I think only occt 7.6 will be left :)

hfiguiere, 2022-05-29:
You forgot issue #61

adrianinsaval, 2022-05-29:
Right! I haven't looked at it yet but I'm no expert in building Qt modules

hfiguiere, 2022-05-29:
The thing is that the current build fix result in a build that crash and the stack trace is more than useless. Trying to get a build with webengine with debug symbols.

adrianinsaval, 2022-05-29:
Is qtwechannel also built? I think that's a new dependency too

hfiguiere, 2022-05-30:
Issue #61 is fix now

hfiguiere, 2022-06-15:
0.20 is on beta now. Will move down to stable soon.

PR #84 (2022-06-15): [beta] FreeCAD 0.20
This is FreeCAD 0.20

Close #80 (it is included) 
Close #2
Close #76 

I'll merge to master later, let's give beta a try.