flathub/org.freecadweb.FreeCAD #85: Possible workflow to deply FreeCAD in corporate settings? (using Flatpak's `create-usb`)

luzpaz, 2022-06-14:
https://docs.flatpak.org/en/latest/usb-drives.html

hfiguiere, 2022-06-14:
that's not a Freecad issue.

luzpaz, 2022-06-14:
Just opening this ticket here for the time being, if that's ok?

adrianinsaval, 2022-06-16:
@hfiguiere not fully related to this issue but I didn't want to create a new one, is it possible to have more than one branch in flathub-beta? For context, during the next development cycle there will be two parallel branches getting development and it would be nice to have distributions of both available, otherwise we'll have to decide which one to give priority.

luzpaz, 2024-11-06:
wait so this is still not something that can be done ?

adrianinsaval, 2024-11-07:
what is to be done? this does not describe any issue.