flathub/org.freecadweb.FreeCAD #88: pip missing in Freecad 0.20 flatpak builds

AmeyaPatwardhan, 2022-07-05:
Unable to install Assembly3 in Freecad 0.20 (Flatpak) on Ubuntu 20.04

Upon using the addon manager-> Assembly3->Install,
and clicking yes to install the python dependencies (py-slvs) the following message pops up:

Failed to execute pip, which may be missing from your Python installation. Please ensure your system has pip installed and try again. The failed command was:

/usr/bin/python3 -m pip --version

Continue with installation of Assembly3 anyway?

Freecad version:
FreeCAD - An open source parametric 3D CAD modeler

ID: org.freecadweb.FreeCAD
Ref: app/org.freecadweb.FreeCAD/x86_64/stable
Arch: x86_64
Branch: stable
Version: 0.20.0.29177
License: LGPL-2.1
Origin: flathub
Collection: org.flathub.Stable
Installation: user
Installed: 1.0 GB
Runtime: org.kde.Platform/x86_64/5.15-21.08
Sdk: org.kde.Sdk/x86_64/5.15-21.08

hfiguiere, 2022-07-08:
And where would pip install things?

We could add pv-slvs to the flatpak, but there is no source package on pypi available.

hfiguiere, 2022-07-08:
IMHO the workbench should bundle it and not depend on "system wide" installation. (it's not the only one that require external python modules)

realthunder, 2022-07-08:
`py-slvs` cannot by bundled because of licensing issue. Using system `pip` will only work if the system Python is compatible with the one used to build FreeCAD. So I think the problem should be fixed by bundling `pip` into FreeCAD build.

hfiguiere, 2022-07-09:
I couldn't find a source distribution on pypi.

And what is the licensing problem you are mentioning, because if you can bundle it, I don't think we can either.

realthunder, 2022-07-09:
The pipy package [home](https://pypi.org/project/py-slvs/#:~:text=py_slvs%20is%20a%20Python%20binding,memory%20optimization%20on%20Linux%20build) links to the repo. It is licensed under GPL, while FreeCAD is LGPL. I don't mean to bundle py-slvs. I mean to bundle pip itself, which is what I assume what's missing here. py-slvs will be installed on demand by user explicitly.

hfiguiere, 2022-07-09:
So I will reiterate the question https://github.com/flathub/org.freecadweb.FreeCAD/issues/88#issuecomment-1179136571: **where will pip install this mpdule?**

Just to clarify. The file system is *READ ONLY* inside the sandbox.

And my armchair analysis of licensing is that if you can't bundle `py_slvs` with your workbench, then surely we can't with the Flatpak.

realthunder, 2022-07-09:
I am not sure how `flatpak` works. For `snap`, it will store common data associated with a particular app in `~/snap/<app_name>/common` directory. For user installed pip package, it will be installed in `~/snap/<app_name>/lib/python<version>/site-packages`.

adrianinsaval, 2022-07-10:
I think the add-on manager uses a freecad specific directory for pip packages, if I'm not mistaken they are installed somewhere in $XDG_DATA_HOME

tinywrkb, 2022-07-10:
@AmeyaPatwardhan maybe try [this](https://github.com/flathub/org.qutebrowser.qutebrowser/wiki#install-python-modules-inside-the-sandbox) with the adjustments for the FreeCAD app. You will need to override the `PATH` environment var and set it to `/app/bin:/usr/bin:/var/data/python/bin`.

AmeyaPatwardhan, 2022-07-13:
@tinywrkb tried using the suggested approach. I could not get it to work. However, this may be because of my lack of knowledge of flatpak environment variables, chroot etc.  Is there any way to create a python virtual environment and simply add this to the flatpak path?

PR #92 (2022-08-21): Issue #88 - Provide pip to install dependencies of workbenches

Closes #88
closed 2022-08-21 by hfiguiere