flathub/org.freecadweb.FreeCAD #93: missing pip executable to allow installation of add-ons requiring addiitonal python modules

rlagerweij, 2022-08-24:
Flatpak version of Freecad 20.1

When trying to install InventorLoader (v1.3) via "Tools -> Addon manager" the installation fails because pip is missing to install required python modules.

Error message: "Failed to execute pip, which may be missing from your Python installation. Please ensure your system has pip installed and try again."

Similar issue in relation to assembly3 workbench described here: 
https://forum.freecadweb.org/viewtopic.php?f=4&t=70020

hfiguiere, 2022-08-24:
did you update recently, I fixed #88  a few days ago

hfiguiere, 2022-08-26:
it doesn't fail because pip isn't here. It does fail because it doesn't install it unlike Assembly3. 

I never the pip error you describe.

hfiguiere, 2022-08-26:
[image]

hfiguiere, 2022-08-26:
Also after installing the "SheetMetal Workbench" I still get the error.

hfiguiere, 2022-08-26:
But if I continue anyway it installs the missing packages with `pip` like issue #88 fixed.

hfiguiere, 2022-08-29:
that's what I observed to. I didn't have sheet metal but I installed it.

closed 2022-08-29 by hfiguiere