flathub/org.freecadweb.FreeCAD #95: AutoRecovery does not work

akvadrako, 2022-09-05:
Every time I open a document in FreeCAD, I see an error message like this:

> 16:53:53  Document with the UUID '2d38467b-cbb2-4db3-a735-3fc73eb2855c' already exists, change to 'e2cca819-bbd6-42e1-9ffe-930a332170db'

Because FreeCAD thinks it's a different document, AutoRecovery doesn't work.This seems to be because FreeCAD uses the PID as part of a lock file and that lockfile always has PID=2 in flatpak, so it thinks I'm opening a document with a duplicate UUID instead of the same document.

The lock file is `/home/dev/.var/app/org.freecadweb.FreeCAD/cache/FreeCAD/Cache/FreeCAD_2.lock`.

hfiguiere, 2022-09-05:
this should be an upstream issue. Using PID for that isn't a good idea. the Flatpak being in a container, it uses namespaces for PID.

adrianinsaval, 2023-04-29:
Is this still an issue? if yes please provide full version information

dfrank-a, 2024-01-17:
Still an issue for me. Version 0.21.2.33771

adrianinsaval, 2024-10-30:
> this should be an upstream issue. Using PID for that isn't a good idea. the Flatpak being in a container, it uses namespaces for PID.

what should be used instead of PID then? I believe this is also the root cause of https://github.com/flathub/org.freecad.FreeCAD/issues/4