flathub/org.freecadweb.FreeCAD #105: How to install OpenFOAM?

kevinsmia1939, 2023-01-22:
I want to use the addon CfdOF. But the addon ask for OpenFOAM executable.
Does OpenFOAM need to be include in the FreeCAD flatpak manifest or it can be done separately?

https://www.openfoam.com/

adrianinsaval, 2023-01-23:
I think we would need an openfoam flatpak, IMO it's too heavy to include with FreeCAD

kevinsmia1939, 2023-01-23:
> I think we would need an openfoam flatpak, IMO it's too heavy to include with FreeCAD

How to build openfoam flatpak that will be seen by FreeCAD? Because CfdOf addon ask for openfoam executable directory.

tkontogi, 2023-12-30:
FreeCAD and CFDof can access an rpm installed openfoam dir, via /run/host dir using default flatpack permissions.

In my case (Fedora 39) I access the openfoam dir via `/run/host/usr/lib/openfoam/openfoam2306/`

[image]

hfiguiere, 2023-12-30:
> FreeCAD and CFDof can access an rpm installed openfoam dir, via /run/host dir using default flatpack permissions.

if this work, it's a coincidence. FreeCAD has already too broad permissions with `host`.

tkontogi, 2023-12-31:
Still under that setup of mine, there are things missing.
For example the wmake is not found by FreeCAD, though openfoam-devel is installed, and wmake is in the path.