flathub/org.freecadweb.FreeCAD #107: Change of domain

neildarlow, 2023-02-04:
FreeCAD has transitioned from freecadweb.org to freecad.org (redirects are in place from freecadweb.org to freecad.org) and users are requested to use the freecad.org domain for accessing online resources. Perhaps this project should be renamed to be consistent with this change?

hfiguiere, 2023-02-04:
no. this is way too disruptive. because that mean freecad needs to be resubmitted and without user intervention all to workebenches and other will not be available.

neildarlow, 2023-02-04:
Didn't KiCad go through a similar transition and that was handled without user intervention?

hfiguiere, 2023-02-04:
I don't know I don't handle KiCAD.

luzpaz, 2023-02-04:
@kkremitzki heads-up

adrianinsaval, 2023-02-04:
looks like kicad indeed did this, but I don't know if it didn't require user intervention for porting over settings and addons
https://github.com/flathub/org.kicad_pcb.KiCad/pull/58
https://github.com/flathub/flathub/pull/2041#issuecomment-758154069

kicad developers preferred the use of the kicad.org domain, up until now freecad didn't have as much focus on the freecad.org domain, I mean the freecadweb.org domain still is listed in some places in the code repo.
While I too would prefer this flatpak be named org.freecad.FreeCAD I don't think it's worth screwing over existing users.

luzpaz, 2023-02-04:
JFYI, there is an open issue https://github.com/FreeCAD/FreeCAD/issues/6415

hfiguiere, 2023-05-27:
it's seems that there is an automatic migration of the preferences directory so it could be made less painful than I orignally thought (it's a fact well hidden in the documentatin). The problem involve resubmitting it with a new name, and this is always friction.

adrianinsaval, 2023-05-27:
I'll see about resubmitting the app, can you point me to where in the documentation the automatic migration is mentioned? What do we need to do to get that working?

hfiguiere, 2023-05-28:
this thing is that it is not documented.

The eol-rebase https://github.com/flathub/flathub/wiki/App-Maintenance#end-of-life

It will trigger the migration.
The code https://github.com/flatpak/flatpak/commit/e17b2635d686c8b91ebbc1b42c6c5d8108586ae6

JustALawnGnome7, 2024-04-24:
@adrianinsaval Any progress on this? Once the app has been resubmitted under the new ID, it looks like the process for triggering an EOL rebase on this repo is a two-line change to `flathub.json`:

```json
{
  "only-arches": ["x86_64"]
  "end-of-life": "The application has been renamed to org.freecad.FreeCAD.",
  "end-of-life-rebase": "org.freecad.FreeCAD"
}
```

Implementing the change in this way signals Flatpak to migrate user data from the old package (`org.freecadweb.FreeCAD`) to the new one (`org.freecad.FreeCAD`). For additional information, see the following documentation:

https://docs.flathub.org/docs/for-app-authors/maintenance/#end-of-life-rebase

adrianinsaval, 2024-04-24:
we'll do the migration for the upcoming 1.0 release of freecad, in the current release we are still using org.freecadweb.FreeCAD ourselves. I guess I could submit it already just for the beta channel

hfiguiere, 2024-09-26:
See https://github.com/flathub/org.freecad.FreeCAD and #137

closed 2024-11-19 by adrianinsaval