flathub/org.freecadweb.FreeCAD #112: Automatic python modules documentation doesn't work

ronnystandtke, 2023-04-02:
FreeCAD supports the automatic generation of information about available Python modules, classes and functions of FreeCAD (see [https://wiki.freecad.org/Std_PythonHelp](https://wiki.freecad.org/Std_PythonHelp)).

The documentation can be opened via the menu item "Help -> Automatic python modules documentation" or via the Python console the command "Gui.runCommand('Std_PythonHelp',0)".

This feature works without any issue in the distribution packages but, unfortunately, not in the flatpak version.

luzpaz, 2023-04-02:
What's the error generated ?

ronnystandtke, 2023-04-02:
There is no error message, neither on the command line, nor the report view or the Python console. There is no reaction at all.

adrianinsaval, 2023-04-03:
if you launch freecad from a terminal and try to do this is there any output in the terminal?

adrianinsaval, 2023-04-03:
Also please specify full version information of both the not working version and the working version (Help->About->Copy to clipboard)

ronnystandtke, 2023-04-03:
> if you launch freecad from a terminal and try to do this is there any output in the terminal?

As I already said, there is no output on the command line. I started FreeCAD in the terminal via
`flatpak run org.freecadweb.FreeCAD`

> Also please specify full version information of both the not working version and the working version (Help->About->Copy to clipboard)

**non-working version (flatpak from flathub):**

    OS: KDE Flatpak runtime (KDE/plasma)
    Word size of FreeCAD: 64-bit
    Version: 0.20.2.29603 (Git)
    Build type: Release
    Branch: (HEAD detached at 0.20.2)
    Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
    Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
    Locale: German/Switzerland (de_CH)
    Installed mods: 
      * Assembly4 0.12.5
      * A2plus 0.4.60h
      * parts_library
      * Help 1.0.3
      * ExplodedAssembly
      * ProDarkThemePreferencePack 1.0.0
      * Curves 0.6.8
      * LCInterlocking

**working version (Ubuntu package):**

    OS: Ubuntu 22.10 (KDE/plasma)
    Word size of FreeCAD: 64-bit
    Version: 0.20.1.
    Build type: Release
    Python 3.10.7, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
    Locale: German/Switzerland (de_CH)
    Installed mods: 
      * Assembly4 0.12.3

adrianinsaval, 2024-11-06:
I can't reproduce this issue on flathub-beta, is this still a problem for you?

ronnystandtke, 2024-11-10:
For the stable branch on flathub it is still broken, the version from flathub-beta seems to work!

adrianinsaval, 2024-11-11:
awesome, if everything goes according to plan 1.0 will be in stable in around a week.

closed 2024-11-19 by adrianinsaval