flathub/org.freecadweb.FreeCAD #113: Could anytree Python library be added for Assembly 4 workbench support?

neildarlow, 2023-04-07:
Hi,

The Assembly 4 workbench has an "Export Files" function (accessible from toolbar and Assembly menu) but this requires the anytree python library which isn't currently a dependency for the FreeCAD flatpak.

Without anytree installed, attempts to use those functions present a warning in the Python Console indicating the library is missing. The relevant Assembly 4 source file for this feature (for reference purposes) is "exportFiles.py".

I understand if there's reluctance to include arbitrary dependencies into the flatpak but this does affect Assembly 4 functionality.

TIA,
Neil

luzpaz, 2023-04-08:
Wait, is anytree a prerequisite for A4? If so then it needs to be added via A4 [package.xml](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/package.xml) metadata file that the FreeCAD Addon Manager will reference and auto install. 

The xml tag necessary: https://wiki.freecad.org/Package_Metadata#%3Cdepend%3E

CC @Zolko-123

Zolko-123, 2023-04-08:
Nah, it was added by someone else, I'm going to remove that dependency

adrianinsaval, 2023-04-08:
Install using addon manager: https://forum.freecad.org/viewtopic.php?p=668000#p668000

or wait for it to be removed from asm4

closed 2023-04-08 by adrianinsaval