flathub/org.freecadweb.FreeCAD #114: Can't generate solar diagrams: Ladybug and Pysolar are missing in the flatpak image

tnozicka, 2023-04-09:
Arch workbench has an object called [Arch Site](https://wiki.freecad.org/Arch_Site) that should be able to generate Solar and wind diagrams. Long-term it seems to head toward Ladybug but according to the wiki description, there is a fallback to Pysolar. Given I got the Pysolar error it seems like none of these is present in the flatpak.

```
16:42:59  The pysolar module was not found. Unable to generate solar diagrams
```

```
$ flatpak info org.freecadweb.FreeCAD

FreeCAD - An open source parametric 3D CAD modeler

          ID: org.freecadweb.FreeCAD
         Ref: app/org.freecadweb.FreeCAD/x86_64/stable
        Arch: x86_64
      Branch: stable
     Version: 0.20.2.29603
     License: LGPL-2.1
      Origin: flathub
  Collection: org.flathub.Stable
Installation: system
   Installed: 1,1 GB
     Runtime: org.kde.Platform/x86_64/5.15-22.08
         Sdk: org.kde.Sdk/x86_64/5.15-22.08

      Commit: ff725ea2feff16b83362126763f65cadc1b84f6f496ab7aa5604b1ae5efd4cb5
      Parent: 29ee4f7e4e24ab3d68976377db40a377c26b9d58a1e2e923ab201de72fdd0c2e
     Subject: remove unnecessary org.freecadweb.FreeCAD.Manual extension (e0c2721d)
        Date: 2023-01-25 00:12:46 +0000
```

adrianinsaval, 2023-04-09:
@yorikvanhavre can you comment on this? is this a module that should be shipped with freecad? if yes I'll see about adding it

yorikvanhavre, 2023-04-11:
Indeed one of these two is needed to generate diagrams...

I think pysolar is the easiest because it's pretty lightweight, just a bunch of py files: https://github.com/pingswept/pysolar/tree/master/pysolar

An alternative to packaging could be to migrate the solar diagram functionality to the BIM workbench. Then, additional dependencies such as pysolar could be handled directly by the addon manager...

tnozicka, 2023-05-20:
Thanks for your input! As this has been lingering for a while and there are 2 proposed paths forward - should we start by adding pysolar to the flatpak image? Unless there is any work going on migrating the functionality to the BIM workbench itself, it seems like the easiest path forward.

yorikvanhavre, 2023-05-22:
I've looked at the code again, actually there is not much point in separating the solar diagram code and migrate it to BIM, as in any case the Arch Site works without it. So there would be not much difference.

We could distribute pysolar with FreeCAD, but since pysolar is GPL, this would force us to change the license of the distributed FreeCAD packages. That's a much deeper discussion to have...

hfiguiere, 2023-05-22:
so which one shall be shipped ? pysolar or ladybug?

yorikvanhavre, 2023-05-22:
ladybug is AGPL so in principle it also forces FreeCAD to be distributed as GPL. That's a decision for the  FreeCAD flatpack maintainers I guess...

hfiguiere, 2023-05-22:
The license is a non issue here. The question was which one is better, but pysolar seems to be the simplest dependency.

hfiguiere, 2023-05-22:
There is a test build on #121 that includes pysolar. Not sure how to test it, but if someone can confirm it works, we can merge it.

yorikvanhavre, 2023-05-23:
> pysolar seems to be the simplest dependency

Yes on a pure practicability point of view, pysolar is indeed much easier to package.

hfiguiere, 2023-06-10:
there was no feedback with the test build, I'll assume it works.

PR #121 (2023-06-10): Issue #114 - Added pysolar
Close #114
closed 2023-06-10 by hfiguiere