flathub/org.freecadweb.FreeCAD #119: Add `flatpak` name to FreeCAD core so it displays in the About dialog if built as flatpak

luzpaz, 2023-05-05:
The FreeCAD [About](https://wiki.freecad.org/About) dialog currently will list AppImage or Snap but not flatpak:
 
https://github.com/FreeCAD/FreeCAD/blob/8d1c9c256318cba56e6cdabefad7445fc0badfb0/src/Gui/Splashscreen.cpp#L751-L757

Shall we add it ?

adrianinsaval, 2023-05-05:
We can, I assume those are environment variables, what are the expected contents of the variables?

adrianinsaval, 2023-05-05:
I'm not sure if it's needed though, since in the os info it shows it's running on flatpak, for example:
```yml
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
```

adrianinsaval, 2023-05-05:
we can check for existence of `FLATPAK_ID` env variable if desired

luzpaz, 2023-05-05:
```cpp
    char *appimage = getenv("APPIMAGE");
    if (appimage)
        str << " AppImage";
    char* snap = getenv("SNAP_REVISION");
    if (snap)
        str << " Snap " << snap;
    str << '\n';
```

Aside:
Any idea why the asterisks sometimes on the char and sometimes on the pointer name ? 

`char *appimage` Vs `char* snap`

hfiguiere, 2023-05-05:
> we can check for existence of `FLATPAK_ID` env variable if desired

it's better to check for the `/.flatpak-info` file.

easyw, 2023-05-17:
in Snap it seems there is an additional info like:
`Version: 0.20.2.29177 +426 (Git) Snap 517`
it would be useful also in FlatPack, to allow WB coders to adapt the codein case of Flat or Snap versions.
I don't know if there is any other way toget this info inside FreeCAD python console

easyw, 2023-05-18:
@luzpaz 
thanks a lot 😄

adrianinsaval, 2024-11-06:
flatpak is already mentioned in the OS field so I consider this unnecessary

luzpaz, 2024-11-06:
@adrianinsaval i disagree... if someone is scanning a list of different About infos in a row having the flatpak text in the same place where snap is helpful and not redudant.

closed 2024-11-11 by adrianinsaval