flathub/org.freecadweb.FreeCAD #120: End of dimension text missing in Sketcher

mfraser, 2023-05-10:
If I open up the Sketcher WB all dimensional text is truncated. 
[image]
If I turn of units in preferences, the text is still truncated. This is only with the Flatpak beta, when I use the weekly AppImage version it looks OK.

[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33038 (Git)
Build type: Release
Branch: (HEAD detached at b3b30e9)
Hash: b3b30e951b46e4dbfe0643934a0b27fb59e9bf32
Python 3.10.6, Qt 5.15.9, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.2.63
  * freecad.gears 1.0.0
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * fcgear 1.0.0
  * Lithophane
  * Curves 0.6.8
  * Assembly4 0.12.4
  * kicadStepUp
Mod 10.18.8
  * CurvedShapes 1.0.4
  * Defeaturing 1.2.0
  * DynamicData 2.46.0
  * fasteners 0.4.56
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.85.0
  * VendorParts
[/code]

adrianinsaval, 2023-05-11:
can you upload your config? Are you using a hidpi monitor? do you have any scaling configured? Does it happen on the stable flatpak aswell?

mfraser, 2023-05-15:
> can you upload your config? Are you using a hidpi monitor? do you have any scaling configured? Does it happen on the stable flatpak aswell?

I am not using an HiDPI monitor and I don't think I have any scaling configured. It also happens with the stable version.

Which files do you want uploaded? FreeCAD.conf, system.cfg & user.cfg?

mfraser, 2023-05-18:
Had a Nvidia update via flatpack a couple of days ago and the issue seems to have gone.

closed 2023-05-18 by mfraser