flathub/org.freecadweb.FreeCAD #127: compressed .wrl (.wrz) files silently fail to decompress

evils, 2023-09-25:
first encountered in the Nixpkgs package,
i had initially opened a [FreeCAD issue](https://github.com/FreeCAD/FreeCAD/issues/10803) because a friend of mine reproduced the problem on linux mint with flatpak
however there we failed to reproduce it on windows, ubuntu and in the appimage
therefore i filed a [Nixpkgs issue](https://github.com/NixOS/nixpkgs/issues/257313)
and am opening this issue

the issues linked above have more details and outline steps to reproduce the problem

a similar issue occurs in the Nix package for KiCad, but not in the flatpak package for KiCad

hfiguiere, 2023-09-25:
Where can we find such a file to try?

evils, 2023-09-25:
any .wrl or .wrz should (not) work
i linked [to this file which i've been using](https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/0a94ebca5a5523db9f6474c262ec3c72ea0241bc/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x01_P2.54mm_Horizontal.wrl?inline=false) in the other issues

hfiguiere, 2024-01-18:
Oh that's a fun one. 

If I start FreeCAD with `--devel` (this loosen the snadbox to allow a debugger or strace), then it works. 

I think we should disable the dynamic loading

adrianinsaval, 2024-11-06:
I see this in terminal when trying to do this>
```
Coin error in zlib glue(): Unable to load zlib DLL/shared object.
Coin warning in SoInput::setBuffer(): Buffer seems to be in gzip format, but zlib support is not available.
Coin read error: Erroneous character(s) after end of scene graph: "". This message will only be shown once for this file, but more errors might be present
	Occurred at line   1 in <memory>
```
are we building coin wrong?