flathub/org.freecadweb.FreeCAD #133: FreeCAD does not start anymore (main window is not shown?)

JNKL, 2024-02-22:
The small startup screen and loading messages are shown but the window does not show up.

After "export -n SESSION_MANAGER" the last Qt-message vanished but window still doesn't show up.

System is default Gnome-install of Debian 12, up-to-date.
```
flatpak -v run org.freecadweb.FreeCAD
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/XXX/.local/share/flatpak
F: Opening user flatpak installation at path /home/XXX/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/XXX/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.kde.Platform/x86_64/5.15-23.08/ba2a33ac552c8ea85a99563f41f3a4b09a542c6b972b1a0125e11e2a731a045b/files/lib32 does not exist
F: Cleaning up unused container id 881751840
F: Cleaning up per-app-ID state for org.freecadweb.FreeCAD
F: Allocated instance id 2801906894
F: Add defaults in dir /org/freecadweb/FreeCAD/
F: Add locks in dir /org/freecadweb/FreeCAD/
F: Allowing dri access
F: Allowing host-fs access
F: Not sharing "/run/media" with sandbox: Unable to open path "/run/media": Datei oder Verzeichnis nicht gefunden
F: Not sharing "/run/spnav.sock" with sandbox: Unable to open path "/run/spnav.sock": Datei oder Verzeichnis nicht gefunden
F: Allowing x11 access
F: Running 'bwrap --args 39 xdg-dbus-proxy --args=42'
F: Running 'bwrap --args 39 FreeCAD'
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Gtk-Message: 22:52:30.105: Failed to load module "canberra-gtk-module"
Gtk-Message: 22:52:30.107: Failed to load module "canberra-gtk-module"
Qt: Session management error: Could not open network socket
```

hfiguiere, 2024-02-22:
what GPU / driver?

JNKL, 2024-02-23:
Thinkpad T480/Intel UHD 620

hfiguiere, 2024-02-24:
I have this 00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)

So it's possibly not the problem.

JNKL, 2024-02-24:
Do you also get
`Gtk-Message: 19:28:04.402: Failed to load module "canberra-gtk-module"`
when running
`flatpak -v run org.freecadweb.FreeCAD`?

hfiguiere, 2024-02-24:
these message are stupidity from the stack, they appear everywhere.

JNKL, 2024-02-24:
Any idea how debug this problem?

`flatpak repair` did not solve the problem.

`flatpak -v run --command=sh --devel org.freecadweb.FreeCAD` exits after some seconds as well. That's strange, right?

hfiguiere, 2024-02-24:
if that happen, then there is something else. Anything in `journalctl -r` ?

JNKL, 2024-02-27:
```
USER@HOSTNAME:~$ flatpak -v run --command=sh --devel org.freecadweb.FreeCAD
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/USER/.local/share/flatpak
F: Opening user flatpak installation at path /home/USER/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/USER/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.kde.Sdk/x86_64/5.15-23.08/b585b0c3e608062b69604c51412483f0955db763db10c347e83e877175c94b3c/files/lib32 does not exist
F: Cleaning up unused container id 155908743
F: Cleaning up per-app-ID state for org.freecadweb.FreeCAD
F: Allocated instance id 3275823853
F: Add defaults in dir /org/freecadweb/FreeCAD/
F: Add locks in dir /org/freecadweb/FreeCAD/
F: Allowing dri access
F: Allowing host-fs access
F: Not sharing "/run/media" with sandbox: Unable to open path "/run/media": Datei oder Verzeichnis nicht gefunden
F: Not sharing "/run/spnav.sock" with sandbox: Unable to open path "/run/spnav.sock": Datei oder Verzeichnis nicht gefunden
F: Allowing x11 access
F: Running 'bwrap --args 39 xdg-dbus-proxy --args=42'
F: Running 'bwrap --args 39 sh'
[📦 org.freecadweb.FreeCAD ~]$ USER@HOSTNAME:~$
```

gives just

```
Feb 27 18:40:13 stockholm systemd[2029]: Started app-flatpak-org.freecadweb.FreeCAD-4114.scope.
Feb 27 18:40:11 stockholm systemd[2029]: Started vte-spawn-6eae9caa-8b02-4cd9-b2ed-2613685061a1.scope - VTE child process 4111 launched by gnome-terminal-server process 3075.
```

The command just exits without any user interaction. Any idea?

JNKL, 2024-03-03:
So I think I solved the problem.
FreeCAD and FreeCAD --devel exited with exit code 137, which means it received SIGKILL.

Giving it the permission to run in background solved the problem.

Don't know what the underlying problem is.

adrianinsaval, 2024-10-30:
this sounds like a portal issue rather than a freecad issue, ok to close?

or do we need to do something on our end to give freecad background permission by default @hfiguiere ?

JNKL, 2024-10-31:
Yes. I think it can be closed. Don't know how I messed this up. Sorry and thank you!

closed 2024-11-01 by adrianinsaval