flathub/org.freecadweb.FreeCAD #134: 404 error for med-3.3.1 library tarball

bestlinuxgamers, 2024-05-04:
Currently the Manifest file can't be built because the `med-3.3.1` source tarball has been removed by the providing website.

I feel like I searched half the internet for the git repository of this library, or at least a reupload of the old 3.3.1 version, and I just can't find it. Can anyone explain to me what open source library just provides periodical source tarballs and no git repository? Or did I miss anything?

Generally I think this Manifest should use fewer source tarballs. They just encourage these problems and they are generally less trustworthy (just look at the recent XZ backdoor).

hfiguiere, 2024-05-04:
> Generally I think this Manifest should use fewer source tarballs. They just encourage these problems and they are generally less trustworthy (just look at the recent XZ backdoor).

this is irrelevant.

bestlinuxgamers, 2024-05-04:
> > Generally I think this Manifest should use fewer source tarballs. They just encourage these problems and they are generally less trustworthy (just look at the recent XZ backdoor).
> 
> this is irrelevant.

I agree that this is not relevant for this issue. But it is certainly relevant for the project in general. I can create a separate issue for that if you want.

peacememories, 2024-10-05:
I just ran into this issue when trying to build the manifest. It's not just `med-3.3.1.tar.bz2` being removed. When directly accessing any link (even the one provided on the salome website) I get a permission error. `files.salome-platform.org` checks the referrer to make "sure" the link was visited on their website :upside_down_face:

hfiguiere, 2024-10-05:
It's fixed in the new package (renamed) https://github.com/flathub/org.freecad.FreeCAD/blob/beta/org.freecad.FreeCAD.yaml#L179

commit e870fdd7 (2024-10-05): Fix med upstream

Upstream changed

Close #134
See also #138
commit 94843bf0 (2024-10-05): Fix med upstream

Upstream changed
Updated HDF5 and netcdf as required.

Close #134
See also #138
commit 1ebecb8a (2024-10-30): Fix med upstream

Upstream changed
Updated HDF5 and netcdf as required.

Close #134
See also #138
PR #139 (2024-11-19): (stable) EOL and fix build
Close #134 
Close #138
closed 2024-11-19 by adrianinsaval