flathub/org.freecadweb.FreeCAD #136: Preparing for 1.0 release

luzpaz, 2024-09-03:
Similar to https://github.com/FreeCAD/FreeCAD-snap/issues/115 this ticket is tracking the roadmap to FreeCAD flatpak v1.0.

- [ ] \<Issue that needs to be addressed\> etc...
- [ ]

luzpaz, 2024-09-14:
Hi @hfiguiere any chance you have some time to help with this ?

adrianinsaval, 2024-09-14:
I'm working on 1.0RC1 now, however this should get published as org.freecad.FreeCAD

hfiguiere, 2024-09-14:
you have to resubmit for that. We can't rename packages lik that.

adrianinsaval, 2024-09-14:
yes, that's the plan

adrianinsaval, 2024-09-23:
worked on at https://github.com/flathub/flathub/pull/5613

hfiguiere, 2024-09-26:
it's on flathub beta with the new app-id.

When this is merged https://github.com/flathub/org.freecadweb.FreeCAD/pull/137 there will be a transition...

closed 2024-11-19 by adrianinsaval