flathub/org.freecadweb.FreeCAD #138: Flatpak build issues

peterwwillis, 2024-09-29:
Hi there, I'm trying to build this Flatpak on Asahi Linux so I can run it on ARM64.

I noticed these two problems:

1. The "libmed" library doesn't seem to download anymore; the upstream vendor appears to be using some anti-download measures. I was able to find an old copy of med-3.3.1 and continue, but I would suggest using a mirror or vendoring it.

2. The build surprisingly moved along for quite a while, but failed on IfcOpenShell:

```
========================================================================
Building module IfcOpenShell in /home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1
========================================================================
Note: switching to 'v0.7.0a2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at a11b51c Revert "Temporarily disable release tag guard"
Cloning into '/home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1/test/input'...
Submodule path 'test/input': checked out 'eca4a52d7c758a92bad6910fa830436a836745bf'
Cloning into '/home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1/src/ifcopenshell-python/ifcopenshell/mvd'...
Submodule path 'src/ifcopenshell-python/ifcopenshell/mvd': checked out '83c12fa494b9d6a5a370d8525dc1f76aa5c89d8d'
Cloning into '/home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1/src/svgfill'...
Submodule path 'src/svgfill': checked out 'f9102b1e9f58e2c3bf9923c1eafa0373eeb872f5'
Cloning into '/home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1/src/svgfill/3rdparty/svgpp'...
Submodule path '3rdparty/svgpp': checked out '536233cbf232c26a5a2d532b63b2c9ab7338bdc0'
Cloning into '/home/peterw/git/org.freecadweb.FreeCAD/.flatpak-builder/build/IfcOpenShell-1/src/ifcopenshell-python/test/Sample-BIM-Files'...
Submodule path 'src/ifcopenshell-python/test/Sample-BIM-Files': checked out '2daf6c46ed9fc7a17259ef6d4c17c3553ef642cd'
Applying patch IfcOpenShell-hdf5.patch
patching file cmake/CMakeLists.txt
Hunk #2 succeeded at 456 with fuzz 2.
CMake Deprecation Warning at CMakeLists.txt:20 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:28 (cmake_policy):
  The OLD behavior for policy CMP0078 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- BINDIR: bin
-- INCLUDEDIR: include
-- LIBDIR: lib
-- Found LibXml2: /usr/lib/aarch64-linux-gnu/libxml2.so (found version "2.11.8")
-- Found Boost: /app/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: system program_options regex thread date_time
-- Boost include files found in /app/include
-- Boost libraries found in /app/lib
-- Looking for Open CASCADE include files in: /app/include/opencascade
-- Looking for Open CASCADE library files in: /app/lib
-- Required Open Cascade Library files found
-- Found Threads: TRUE
-- Found SWIG: /app/bin/swig (found version "4.0.2")
CMake Warning (dev) at /run/build/IfcOpenShell/src/ifcwrap/CMakeLists.txt:38 (FIND_PACKAGE):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.11.so (found version "3.11.9")
CMake Warning (dev) at /run/build/IfcOpenShell/src/ifcwrap/CMakeLists.txt:69 (FIND_PACKAGE):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3.11 (found version "3.11.9")
CMake Deprecation Warning at /run/build/IfcOpenShell/src/svgfill/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning (dev) at /run/build/IfcOpenShell/src/svgfill/CMakeLists.txt:6 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for CGAL include files in: /app/include
-- Found Boost: /app/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")
-- Boost include files found in /app/include
-- Configuring done (0.6s)
-- Generating done (0.2s)
-- Build files have been written to: /run/build/IfcOpenShell/cmake/_flatpak_build
[7/137] Building CXX object CMakeFiles/IfcParse.dir/run/build/IfcOpenShell/src/ifcparse/parse_ifcxml.cpp.o
/run/build/IfcOpenShell/src/ifcparse/parse_ifcxml.cpp: In function ‘IfcParse::IfcFile* IfcParse::parse_ifcxml(const std::string&)’:
/run/build/IfcOpenShell/src/ifcparse/parse_ifcxml.cpp:654:28: warning: ‘int xmlSAXUserParseFile(xmlSAXHandlerPtr, void*, const char*)’ is deprecated [-Wdeprecated-declarations]
  654 |         xmlSAXUserParseFile(&handler, &state, filename.c_str());
      |         ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /run/build/IfcOpenShell/src/ifcparse/parse_ifcxml.cpp:7:
/usr/include/libxml2/libxml/parser.h:889:17: note: declared here
  889 |                 xmlSAXUserParseFile     (xmlSAXHandlerPtr sax,
      |                 ^~~~~~~~~~~~~~~~~~~
[15/137] Building CXX object CMakeFiles/IfcParse.dir/run/build/IfcOpenShell/src/ifcparse/IfcHierarchyHelper.cpp.o
In file included from /run/build/IfcOpenShell/src/ifcparse/IfcHierarchyHelper.cpp:30:
/run/build/IfcOpenShell/src/ifcparse/../ifcparse/IfcHierarchyHelper.h:253:14: warning: ‘void {anonymous}::set_children_of_relation(IfcUtil::IfcBaseClass*, aggregate_of_instance::ptr&)’ defined but not used [-Wunused-function]
  253 |         void set_children_of_relation(IfcUtil::IfcBaseClass* t, aggregate_of_instance::ptr& cs) {
      |              ^~~~~~~~~~~~~~~~~~~~~~~~
/run/build/IfcOpenShell/src/ifcparse/../ifcparse/IfcHierarchyHelper.h:247:36: warning: ‘aggregate_of_instance::ptr {anonymous}::get_children_of_relation(IfcUtil::IfcBaseClass*)’ defined but not used [-Wunused-function]
  247 |         aggregate_of_instance::ptr get_children_of_relation(IfcUtil::IfcBaseClass* t) {
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~
[19/137] Building CXX object CMakeFiles/IfcGeom_ifc2x3.dir/run/build/IfcOpenShell/src/ifcgeom/IfcGeomFaces.cpp.oError: module IfcOpenShell: Child process killed by signal 15
```

SagarChandra07, 2024-09-29:
try this

https://mega.co.nz/#!qq4nATTK!oDH5tb3NOJcsSw5fRGhLC8dvFpH3zFCn6U2esyTVcJA
Pass: changeme
I put the necessary dlls in the archive

SagarChandra07, 2024-09-29:
try this

https://mega.co.nz/#!qq4nATTK!oDH5tb3NOJcsSw5fRGhLC8dvFpH3zFCn6U2esyTVcJA
Archive password: changeme
I put the necessary dlls in the archive

hfiguiere, 2024-09-29:
and there is an aarch4 build

closed 2024-09-29 by hfiguiere

peterwwillis, 2024-09-29:
@hfiguiere thanks, the website linked from that repo 404s, but I was able to add the beta remote and install it from there. Cheers!

Also I got malware spam'd by a user @SagarChandra07 in this issue, it seems they deleted their comments right after adding them. I hope @github fixes this :(

hfiguiere, 2024-09-29:
I did delete the comemnts, and blocked the user.

PR #139 (2024-11-19): (stable) EOL and fix build
Close #134 
Close #138